tamberg/fhnw-iot #1: Feather Huzzah ESP8266 resets if used with Particle Grove adapter

tamberg, 2019-02-19:
The Feather Huzzah ESP8266 board seems to reset if used with Particle Grove adapter.

The board keeps rebooting after a short while (check the serial output).

I'm still investigating the issue, maybe a pin pulled to GND.

(The issue only applies to this board.)

tamberg, 2019-02-22:
> My Huzzah board keeps crashing and resetting, whats up with that?

> The most common reason for crashes is power failure. Make sure you're powering the Huzzah with a good ~5V power supply, and if you're using a USB-Serial cable, that it's plugged into the mainboard of your computer or through a powered hub!

https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/faq#faq-2

(Not sure, via USB-C it should get enough power.)

tamberg, 2019-02-26:
https://community.particle.io/t/particle-mesh-starter-kit-grove-adapter-issue/48059

tamberg, 2019-02-26:
https://community.particle.io/uploads/default/original/3X/f/5/f53141349fdb736887dbd6e5bc55586220be32ba.png via https://community.particle.io/t/particle-mesh-starter-kit-grove-adapter-issue/48059/2

okaerin, 2019-05-15:
According to adafruit the CH_PD pin needs to be pulled high or else the board resets. This is violated by the particle mesh board as it connects the pin to GND. It can be solved by bending the CH_PD pin away
https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/pinouts/

tamberg, 2019-05-15:
Very nice, thanks! Then the fix is to (slowly) bend the CHPD pin towards the inside as shown here:
[image]

closed 2019-05-15 by tamberg

tamberg, 2019-07-17:
Here's a better fix: 
[image]

tamberg, 2019-07-17:
And a pragmatic fix, which consumes a bit more energy, using a jumper cable:
[image]