tamberg/fhnw-iot #5: DHT11 returning with CPU Speed not suppored

sabrand, 2019-02-25:
Die DHT Library unterstützt den nRF52840 noch nicht, der Prozessor ist zu schnell mit 64MHz.

Um die CPU des nRF52840 zu unterstützen, muss die DHT Library angepasst werden. Im File DHT.h müssen die folgenden Zeilen eingefügt werden:

// 64MHz nRF52840  <== UNSER Controller
#elif (F_CPU == 64000000UL)
#define COUNT 20

am besten direkt nach den 2 Zeilen für den 48MHz Eintrag.
Der Wert für COUNT ist experimental.

closed 2019-02-26 by tamberg

tamberg, 2019-02-26:
Updated the Wiki https://github.com/tamberg/fhnw-iot/wiki/Grove-Sensors

okaerin, 2019-03-18:
Ist eine bessere Lösung zum DHT-Problem:
//sensors
#define DHTCOUNT 20 // experimental setting
#define DHTPIN 2     // what pin we're connected to
DHT dht(DHTPIN, DHTTYPE,DHTCOUNT);