tamberg/fhnw-iot #11: 4-Digit Display does not seem to work with nRF52840

tamberg, 2019-09-30:


GabrielMassaroFHNW, 2019-09-30:
Klappt bei mir mit dem nRF:

**Libraries:** 

- Grove 4-Digit Display (by Seed Studio)
- TimerOne (by Jesse Tane)

**Code:** 
Example Code -> Grove 4-Digit Display -> Number Flow
oder 
https://github.com/fhnw-iot-5ibb1/fhnw-iot-work-02-GabrielMassaroFHNW/blob/master/02_Kitchentimer/02_Kitchentimer.ino

tamberg, 2019-09-30:
Confirmed. The example _File > Examples > Grove 4-Digit Display -> NumberFlow_ works on the nRF52840 with the following pin settings:

```
#include "TM1637.h"
#define CLK 9 // Grove adapter D4
#define DIO 10

...
```

tamberg, 2019-09-30:
Adapted Wiki https://github.com/tamberg/fhnw-iot/wiki/Grove-Actuators#4-digit-display-tm1637 > Code > Arduino

closed 2019-09-30 by tamberg